Amertz08/Robot #29: Create basic web app

Amertz08, 2018-01-20:
- Flask app that has 1 route that returns "hello world"
- Dockerfile with python 3.6 as base
- Dockerfile should install all necessary packages
- Dockerfile should expose necessary ports
- Dockefile should have proper command to keep container running on start
- Container should wait for all necessary previous container to start
- Add `web` service to `docker-compose.yml` that will bind local code base to allow hot development

Hz-Huang, 2018-01-26:
Added Dockerfile & basic flask app

Amertz08, 2018-01-26:
Cool this looks basically g2g. Can you submit a PR for your branch so I can add a couple comments on changes.

Amertz08, 2018-01-26:
Wait nvm. You still need to `docker-compose` which you noted in your meeting notes. I'll just make my comments here. Expose and run the web app on port `5000` as `80` is the default port for actual web servers such as NGINX. Also on your `pip install` specific a specific Flask version. In our case `0.12.0` will be fine.

PR #50 (2018-02-01): 29 web app
Resolves #29
commit 0a269651 (2018-02-01): Merge pull request #50 from Amertz08/29-Web-App

Resolves #29
closed 2018-02-01 by Amertz08