Amertz08/Robot #30: Create basic route solving

cduddy, 2018-01-20:
Create a basic route solver given a 2-d map.

Amertz08, 2018-01-20:
Vague. Elaborate on requirements

cduddy, 2018-01-20:
Updated. I just added it to assign real quick so I could push up for you guys.

Amertz08, 2018-01-20:
Is this solution based on previous solutions you found in research? I think its best we don't reinvent the wheel. This issue is more broadly defined as navigation. We don't know how we are actually going to navigate this thing. My thought would be to find something that works and tweak as needed.

cduddy, 2018-01-23:
I've been doing research over the weekend trying to pin point good open source solutions for this, I've found a few that may work well for us. Should we decide to make our own, we should use a breadth-first search algorithm to solve for the shortest path. I'm pinpointing a few open source solutions that we can use now.

cduddy, 2018-01-23:
This one looks very promising https://github.com/citiususc/hipster

Amertz08, 2018-01-23:
I meant actual research like academic papers that view it from a conceptual and systems point of view on building a line following robot for warehouses. That is a java based library and we are planning on using Python.

cduddy, 2018-01-23:
Adam, we're in a very specific use case here, there aren't scholarly papers written on "How to do route selection for your factory robot" . I have a solution planned with Nick for the route detection/ driving instructions between the arduino and the pi. Regardless, this is a simple _python_ guide to doing graph theory and path detection : https://www.python-course.eu/graphs_python.php . We can explain what we have planned tomorrow.

cduddy, 2018-02-02:
@Amertz08 can we close this? The baseline is done.

cduddy, 2018-02-27:
Baseline is done for this. Improvements have been made. Closing.

closed 2018-02-27 by cduddy