Amertz08/Robot #36: Convert ints to uint8_t on sensor lib

Amertz08, 2018-01-26:
Convert all `int` types to `uint8_t` in `sensor_smoothing` lib per the following

https://arduino.stackexchange.com/questions/30749/int-vs-uint8-t-vs-uint16-t

nBukaty, 2018-01-26:
8 bits is not large enough. The analogueRead() function gives you something
like a 10 bit value.

Amertz08, 2018-01-26:
Ah I was unaware of that. Well then use  `int` for that. But still think it is in our best interest to use `uint8_t` whenever possible to limit memory usage and CPU clocks

Amertz08, 2018-01-26:
that or `uint16_t` as I think using `int` might add a slight amount of overhead

Amertz08, 2018-01-26:
Just use `int` for any value > 255 but we should be using `uint8_t` for any value < 255 due to limited memory and CPU cycles on these boards